RoganDawes/LOGITacker #4: Patch for APR Brother Dongle

nieldk, 2019-07-12:
Here is a patch, to make LOGITacker work with the APR Brother Dongle (based on the pca10059 build)

aprdongle.patch

-```
-- logitacker_pca10059.ld	2019-07-12 07:41:20.386670177 +0000
+++ logitacker_pca10059a.ld	2019-07-12 07:41:46.526489027 +0000
@@ -5,8 +5,8 @@
 
 MEMORY
 {
-  FLASH (rx) : ORIGIN = 0x1000, LENGTH = 0xff000
-  RAM (rwx) :  ORIGIN = 0x20000008, LENGTH = 0x3fff8
+  FLASH (rx) : ORIGIN = 0x26000, LENGTH = 0xda000
+  RAM (rwx) :  ORIGIN = 0x200022e0, LENGTH = 0x3dd20
 }
 
 SECTIONS
```

mame82, 2019-07-12:
Do you think it is possible to do a UF2 conversion (assume this could only be flashed with SWD debug probe)

nieldk, 2019-07-12:
Yes, UF2 works fine. 

`uf2conv -c -f 0xADA52840 logitacker_pca10059.hex
`
will generate flash.uf2 that can be copied to the removable drive.
Remove and reinsert after flashing

mame82, 2019-07-12:
Will add it to release, as soon as shipping issues are solved.

Awesome work on this, thanks!

RoganDawes, 2019-07-12:
Struggling to make out the difference in the patch, can you use code tags, please? Or attach as a file?

nieldk, 2019-07-12:
> Struggling to make out the difference in the patch, can you use code tags, please? Or attach as a file?

done

RoganDawes, 2019-07-12:
If creating a new LD script for this, I guess I would name it something like ```nrf52840_uf2.ld```, sounds like it should be applicable to any device using the UF2 bootloader on an nRF52840. e.g. I could probably flash this to my actual pca10059 which has a UF2 bootloader installed (via SWD), but also to an AprilBrother dongle, etc.

nieldk, 2019-07-12:
> If creating a new LD script for this, I guess I would name it something like `nrf52840_uf2.ld`, sounds like it should be applicable to any device using the UF2 bootloader on an nRF52840. e.g. I [...]

Yes, its easy to make a new build like that, just copy the exisiting pca10059 folder, change the pca10059 namings in Makefile and you are good to go.
I have attached here a copy of this repository, which contains the changes to build "apr-dongle"
[LOGITacker-master.zip](https://github.com/mame82/LOGITacker/files/3386378/LOGITacker-master.zip)

closed 2019-07-13 by nieldk