RoganDawes/LOGITacker #9: Revision command

grymoire, 2019-08-17:
It would be nice to have a command or ability that tells us the revision of the firmware installed. This will be especially useful to those who aren't compiling/installing from source

nieldk, 2019-08-17:
you mean, like 'version', this is already there

`LOGITacker (discover) $ version                                                 
LOGITacker by MaMe82                                                            
Version: v0.1.3-beta`

mame82, 2019-08-17:
My bad, have to update the README.

LIGHTSPEED mode for G-Series receivers isn't documented either.

Sorry for the inconvenience, currently too busy

grymoire, 2019-08-19:
No problem, Marcus. It would also have a "help" or "?" command that lists the commands. I would have found out about the "version" command that way.  Cheers! and thanks for this fantastic project.

closed 2019-09-02 by mame82