RoganDawes/LOGITacker #10: build/logitacker_aprdongle.uf2 version v0.1.4-beta appears to be corrupted

fraabye, 2019-08-23:
This binary does not appear to be working with the April Brother dongle as the v0.1.3 did.
Steps to reproduce: 
1. Connect April Brother nRF52840 Dongle as removable drive NRF52BOOT.
2. Copy logitacker_aprdongle.uf2 to the mounted drive. 
3. Expected outcome, a disconnected device / reflash, does not happen and the drive NRF52BOOT drive stays mounted. 

The dongle has been tested successfully with the v0.1.3 build afterwards.

mame82, 2019-08-23:
I re-tested the firmware, just now, and can't reproduce the issue.

File used was the on this one: https://github.com/mame82/LOGITacker/releases/download/v0.1.4-beta/logitacker_aprdongle.uf2

The firmware in use before flashing v0.1.4 was v0.1.3, too.

mame82, 2019-08-23:
Additional note: The dongle fails to reboot after flashing.

Behavior observed:

1) Put dongle in bootloader mode (hold button while plugging to host)
2) Copy firmware to `NRF52BBOT`
3) Dongle LED starts blinking while flashing
4) `NRF52BOOT` mount disappears and dongle stops blinking (flashing finished), but doesn't reboot
5) After re-plug of the dongle, LOGITacker boots as intended

mame82, 2019-08-23:
Question:

Could it be that you accidentally copied the `logitacker_aprdongle.hex` file instead of the UF2 one, from the build directory?

fraabye, 2019-08-24:
It turned out to be a proxy mangling the firmware during download. After r edownloading it (again) using a direct connection it works as expected. Sorry for the inconvenience!

closed 2019-08-24 by fraabye