RoganDawes/LOGITacker #15: Feature Request : French Azerty Support

leperjulien, 2019-09-03:
Hi,

Can't wait to play with this ! I will wait for tommorow to detach the pcb from the dongle and solder it to my USB connector from adafruit that was laying around, but while looking at the source code i saw that only US and DE were present, so i'd like to take care of adding support for french (azerty) keyboard since you must have a lot of work already and i'd hate to add more :)

I just want to be sure to do the right thing before i start so i don't spend time on it for nothing, i was planning on using this file as reference : 

[https://raw.githubusercontent.com/mame82/UnifyingVulnsDisclosureRepo/master/tools/mjackit/keymaps/fr.json](https://raw.githubusercontent.com/mame82/UnifyingVulnsDisclosureRepo/master/tools/mjackit/keymaps/fr.json)

to modify this one (starting at line 275) : 

[https://raw.githubusercontent.com/mame82/LOGITacker/master/logitacker/logitacker_keyboard_map.h](https://raw.githubusercontent.com/mame82/LOGITacker/master/logitacker/logitacker_keyboard_map.h)

but maybe i've got it all wrong and this is not the way to do it.. in which case i'll just leave this here and hope it can be implemented in the future :)

leperjulien, 2019-09-12:
So here is what i was able to come up with : 
https://github.com/leperjulien/LOGITacker/blob/master/logitacker/logitacker_keyboard_map.h

Now i just need to figure out how to compile it into a valid uf2 file 😇

Mera-balou, 2019-10-08:
Hey @leperjulien, i also want to add a language. Did you manage to compile uf2 file with azerty support?

leperjulien, 2019-10-12:
Yes i managed to do it thanks to @mame82 who put me on the right track when i asked him on twitter !

you can find the compiled firmware in the fork of LOGITacker on my github if you need further help let me know

learningsomethingnewanduseful, 2019-10-23:
@leperjulien i do like to know how did u manage to do it. could you add it here so anyone else who likes to know can find it.

cHJlaXpoZXI, 2019-10-23:
Hi leperjulien,

Thanks for the work for the French language. Can you give the method to compile? @mame82 has released the 2.1 beta version, and I want with the French keyboard compatibility.

Thanks in advance ;)

leperjulien, 2019-10-26:
Sure you need to download this script : [https://raw.githubusercontent.com/microsoft/uf2/master/utils/uf2conv.py](url)

And next you will navigate to : LOGITacker/apr-dongle/blank/armgcc/

Next you want to type : `make` and hit enter ! When the build is finish you should have a new folder named [_build](url).

Put the file named uf2conv.py you download earlier in the same folder as the .hex file you just builded and then use this command to conver the .hex to .uf2 :  

`python uf2conv.py logitacker_aprdongle.hex -c -f 0xADA52840 -o firmware.uf2`

New firmware is now here => firmware.uf2

All the credits for this goes to @mame82 :D

mame82, 2019-11-22:
Guys, here's a commit containing a layout addition for DA.

You might find it useful, as it shows which parts of code have to be touched:

https://github.com/mame82/LOGITacker/commit/f44c0c1cff622fa63c8a9763540216d22a442bba

sogewasp, 2020-01-14:
I've managed to make it compile and work with the added FR language (a lot of work was already done by @leperjulien).

Anyway this line seems to break compilation:
```
// French                                                                                                             

DEF_REPORT_SEQUENCE_RALT_AND_EQUAL(RALT_AND_EQUAL,    »·· DEF_REPORT(HID_MOD_KEY_RIGHT_ALT, HID_KEY_EQUAL));
```

Gives this error:                                                                                                                     
```
../../../logitacker/logitacker_keyboard_map.h:240:37: error: expected ')' before '{' token
 #define DEF_REPORT_1(modifier,key1) { .mod = modifier, .keys = { key1 } }
```

So I've just commented it. I don't think it changes that much as another line is already present from DA config :
```
DEF_REPORT_SEQUENCE_NAMED(US_RALT_AND_EQUAL,»···»···»···         DEF_REPORT(HID_MOD_KEY_RIGHT_ALT, HID_KEY_EQUAL));
``` 

The pull request which adds this feature is  #44

FMehault, 2020-02-22:
This little message to inform you that I just managed to add FR keyboard support on LOGITacker v0.2.3-beta, inspired by @leperjulien's work and with @mame82's indications. Many thanks to them.

I've made some modifications  to the following files :

[https://github.com/FMehault/LOGITacker/blob/LOGITackerFR/logitacker/logitacker_keyboard_map.h](url)
[https://github.com/FMehault/LOGITacker/blob/LOGITackerFR/logitacker/logitacker_keyboard_map.c](url)

The covert channel is functional with "option inject language fr". Enjoy !

FMehault, 2020-03-05:
I compiled the azerty keyboard support for all dongles

[https://github.com/FMehault/LOGITacker/tree/master/build](https://github.com/FMehault/LOGITacker/tree/master/build)