RoganDawes/LOGITacker #17: Nordic nRF52840 Dongle (pca10059) - Unable to store settings/scripts

johnc2k, 2019-09-04:
My nRF52840 no longer wants to store any settings/scripts.
I have tried the last 3 firmwares releases and I encounter the same problem each time:

`LOGITacker (discover) $ options store
<warning> LOGITACKER_OPTIONS: failed to find Flash Data Storage record for global options: 2
<error> LOGITACKER_OPTIONS: failed to write global options to Flash Data Storage
<info> LOGITACKER_OPTIONS: global options stored to Flash Data Storage`

`LOGITacker (discover) $ script store test
<error> LOGITACKER_SCRIPT_ENGINE: failed to write first task for script storage
<info> app: Storing script failed`

Do you have any suggestions on what the problem could be?

Thanks

John

mame82, 2019-09-04:
Sounds like flash layout is corrupted. Maybe you forgot to run the `erase_flash` command

johnc2k, 2019-09-04:
Yep that fixed it, many thanks.

John

closed 2019-09-04 by johnc2k

grymoire, 2023-02-17:
I have the same problem, however, I don't see how to erase the flash memory. I'm using nRFconnect programmer v3.0.6 on Linux, and the "erase all" is disabled. According to [the Nordic DevZone](https://devzone.nordicsemi.com/f/nordic-q-a/55761/nrf-connect-programmer-erase-button-always-disabled), this is done on purpose.