RoganDawes/LOGITacker #18: Feature Request: Active devices active refresh with signal strength

ResistanceIsUseless, 2019-09-07:
It would be nice to see the active devices signal strength when looking as devices. That way it's easier to target devices that are still actively in range.

mame82, 2019-09-07:
If a device it out of range and no frames are received, RSSI can't be updated.

In as "discover" mode is pseudo promiscuous, the probability of missing valid device frames is high. The risk to miss a frame has to be multiplied by the channels used by the respective, as discover only runs at one channel at a single point in time. Additionally, tx intervals of Unifying devices are adaptive, so it gets impossible to find a dwell time per channel, which applies to all devices on air.

This (and other facts) make it almost impossible to track RSSI if every discovered device.

Of course in passive enum mode, those issues don't exist, as the selected device is continuously monitored.

A more valuable approach would be to introduce a "last time spotted" timer. At least I thought about this, but on a microcontroller this adds a bunch of unnecessary overhead (updating state values of a growing device list on regularly basis, instead of extending the list based on a event).

So the short answer: I currently have no idea which fulfills your demand, while keeping computing overhead/memory consumption on a reasonable level.

I leave this request open. Maybe some good ideas arise from the community.

Thx so far

mame82, 2019-09-07:
Add up, just in case I get you wrong:

For passive enum, it should be possible to track RSSI for the monitored device.

If it is updated in the device list, it would have to be called "last known RSSI".