RoganDawes/LOGITacker #19: Supported dongles for this attck??

harshil2105, 2019-09-14:
It's very hard for me to find CU007 OR CU0012 dongles are there any other dongle that has this same vulnerability in LOGITacker? 
Like for example CU0008 Is very common does that dongle also has the same vulnerability?

mame82, 2019-09-14:
Almost all Logitech receivers have vulnerabilities, which ones depends on the firmware version. 

This includes for example CU0007, CU0008, CU0010, CU0012, CU0016

mame82, 2019-09-14:
If you ask for pairing, this is supported for:

- Unifying (cu0007, cu0008, cu0012)
- G-Series Lightspeed (cu0008)
- G700 (cu0007)

Pair sniffing is additionally supported for:

- Spotlight (cu0016)
- R500 (cu0016)

harshil2105, 2019-09-15:
I'm asking because CU0008 IS very common for me to find....
Will CU0008 work exactly the same like CU0007 
I mean the guidelines provided are exactly same for this chip set too?

mame82, 2019-09-15:
It depends? Could you be more clear on your use case?

harshil2105, 2019-09-16:
Yes sure. 
If I check on Amazon I see only CU0008. 

And according to these amazing guides by LucaBongiorni. 

(https://medium.com/@LucaBongiorni/usbsamurai-a-remotely-controlled-malicious-usb-hid-injecting-cable-for-less-than-10-ebf4b81e1d0b )

They use CU0007 
What I'm basically asking is can I Use the same set of instructions for my CU0008 unifying dongle. 

Are the instructions exactly the same for this chip set too (CU0008)

THANKS ANS YOU ARE AMAZING AS ALWAYS!! 

side note:--> Will you ever create a discord server...I can most probably make it for you make channels and everything...add bots etc

mohmmadd, 2024-01-01:
> If you ask for pairing, this is supported for:
> 
> * Unifying (cu0007, cu0008, cu0012)
> * G-Series Lightspeed (cu0008)
> * G700 (cu0007)
> 
> Pair sniffing is additionally supported for:
> 
> * Spotlight (cu0016)
> * R500 (cu0016)

Does pair sniffing works for cu0010?

RoganDawes, 2024-01-01:
> > If you ask for pairing, this is supported for:
> > 
> > * Unifying (cu0007, cu0008, cu0012)
> > * G-Series Lightspeed (cu0008)
> > * G700 (cu0007)
> > 
> > Pair sniffing is additionally supported for:
> > 
> > * Spotlight (cu0016)
> > * R500 (cu0016)
> 
> Does pair sniffing works for cu0010?

Pair sniffing should work for all dongles, as it is an over the air weakness, not a firmware weakness, to my understanding.

closed 2024-01-01 by RoganDawes