RoganDawes/LOGITacker #21: new version problems 0.2.1-beta

y0d4a, 2019-09-19:
hi mame :)
with new version i have issue to execute inject...

```
LOGITacker (injection) $ script load calc_win10
<info> LOGITACKER_SCRIPT_ENGINE: script calc_win10 file_id 1004
<error> LOGITACKER_SCRIPT_ENGINE: logitacker_script_engine_load_script_from_flas                                                                                                                                                                                                                                             h: failed to read task data
<info> app: loading script failed
LOGITacker (injection) $ inject target F3:F4:95:87:08
inject target F3:F4:95:87:08
Trying to send keystrokes using address F3:F4:95:87:08
<info> app: parsed addr len 5:
<info> app:  F3 F4 95 87 08         |.....
<info> LOGITACKER_PROCESSOR_INJECT: Stop injection mode for address F3:F4:95:87:                                                                                                                                                                                                                                             08
<info> LOGITACKER_RADIO: Channel hopping stopped
<info> LOGITACKER_PROCESSOR_INJECT: Initializing injection mode for F3:F4:95:87:                                                                                                                                                                                                                                             08
<info> LOGITACKER_RADIO: Channel hopping stopped
<info> ESB_ILLEGALMOD: Using channel table 'Unifying'
<info> ESB_ILLEGALMOD: New channel table with length 25
LOGITacker (injection) $ inject execute
<info> LOGITACKER_PROCESSOR_INJECT: No more tasks scheduled
<info> LOGITACKER_PROCESSOR_INJECT: script execution succeeded
<info> LOGITACKER: Injection processing resumed
```
but nothing happening...

y0d4a, 2019-09-19:
found it, remove all old scripts and create new one... :)

closed 2019-09-19 by y0d4a

mame82, 2019-09-20:
After update you have to run `erase_flash` once, otherwise more errors are likely to occur