RoganDawes/LOGITacker #22: Issue with April USB Dongle 52840

cHJlaXpoZXI, 2019-09-21:
Hi Mame82,

I cannot find anything for my issue, so I ask you if you have some info. My dongle doesn’t seem to work. I’ve tested on Linux and Windows and nothing appear on USB debug.

I tried your steps to flash Logitacker inside it and cannot have the USB storage for the firmware inside. I tried multiple time plug it with the button pushed or double push before plugin and nothing append.

Do you think my device is bricked?

Sorry for the inconvenience.
Thank you,

Mera-balou, 2019-09-29:
Have you tried this procedure?
https://github.com/mame82/LOGITacker#242-remark-on-bootloader

cHJlaXpoZXI, 2019-10-08:
Hello Mera-balou,

Sorry for the delay. I have tested and I doesn't work either. I have talk with the store where I bought the device, they resend me a new device.

Thanks for your support.

closed 2019-10-08 by cHJlaXpoZXI