RoganDawes/LOGITacker #23: Issue with C-U0008

yiu-hongsum, 2019-09-25:
So I managed to input all my scripts and stuff but when i perform 'inject execute', the scripts are sent and transmission is successful but nothing happened. May I know if logitech actually updates the firmware by itself or am i doing something wrongly?

mame82, 2019-09-27:
It doesn't depend on the Unifying receiver, but on the firmware version, which kinds of vulnerability exists.

Rule of thumb:
Only the oldest Unifying receiver firmwares accept plain injection (MouseJack). But, also the newest firmware accept injection, if the encryption key is known, because

a) pairing of a keyboard was sniffed with LOGITacker (CVE-2019-13052)
b) the key was dumped with `munifying` (receiver is vulnerable to CVE-2019-13055) and manually added to LOGITacker.

For details on patched vulnerabilities across different firmware versions, see the respective READMEs in Logitech's repo (CU0007 is RQR12, CU0008/0016 is RQR24):

https://github.com/Logitech/fw_updates/tree/update2019-08-27/RQR12

antonowicz, 2019-12-27:
@mame82 is there some repo with older versions of firmware (vulnerable) for attack?