RoganDawes/LOGITacker #24: How to use nrfutil.exe command-line tool to make a dfu with precompiled hex file for nRF52840?

Droid-MAX, 2019-09-25:
I just follow this [README](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/README.md), then downloaded nrfutil.exe and execute it.
But it has three firmware types, I am really a little confused. So anyone can give me some help?

Droid-MAX, 2019-09-25:
I want to pack hex to dfu zip package file, but there are some key parameters that are unclear, for example, the SDK version of SoftDevice.

Droid-MAX, 2019-09-27:
[image]

I have a very extreme network review in my area, so normal development work is very difficult to complete. This programmer plugin is stuck in the loading process even if I install it manually.

Droid-MAX, 2019-09-27:
[image]

I even used several proxy tools, and they didn’t help.

mame82, 2019-09-27:
Please file issues related to firmware flashing at support of the vendor 8f the specific board

closed 2019-09-27 by mame82