RoganDawes/LOGITacker #26: hi ,may i know how to re-flash  nRF52840

dskho, 2019-10-05:
i am current use v1.0.4 beta version on nRF52840 ,  wan to re-flash to v0.2.1-beta
but always showing "Unsupported device. The detected device could not be recognized as neither JLink device nor Nordic USB device."
do you have any suggestion ?

[image]

domshia, 2019-11-08:
I'm facing exact same issue, but it has to do with a nRFConnect software :(

NotMedic, 2020-03-01:
It looks like your dongles likely have the new UF2 bootloader. If so, you'll need to put it into "DFU" mode by holding the button when you plug it in and it'll show up as an external disk.  

You'll need to download logitacker_mdk_dongle.hex, use the conversion instructions on the makerdiary nrf52 repo for "DFU via UF2 bootloader," and copy the resulting .uf2 file to the dongle.

The repo:
https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/uf2_bootloader

Saving you a click looking for the uf2conv.py script:
https://github.com/Microsoft/uf2/blob/master/utils/uf2conv.py

And the command:
uf2conv.py logitacker_mdk_dongle.hex -c -f 0xADA52840

And the copy the flash.uf2 file to the external drive.  Once it's copied and the light on the dongle goes out, unplug it and plug it back in, and you should see a new COM port. 

Hope that helps.