RoganDawes/LOGITacker #31: G700 (flashed cu0007) pairing with April Brother doesn't work

bl4ckOut, 2019-11-07:
Hi,
as the title says, pairing a cu0007 dongle flashed with the G700 firmware doesn't work with the April Brother. I didn't know where to put this issue, in the munifying repo or here, so if this is the wrong place please just inform me and I will create the issue in the other repo as well.

LOGITacker Version: 0.2.1
munifying Version: current git status

Log:

> ./munifying info
> Found CU0007 Dongle for G700/G700s mouse
> Using dongle USB config: Configuration 1
> Resetting dongle in order to release it from kernel (connected devices won't be usable)
> EP descr: ep #1 IN (address 0x81) interrupt - undefined usage [8 bytes]
> EP descr: ep #2 IN (address 0x82) interrupt - undefined usage [20 bytes]
> HID++ interface: vid=046d,pid=c531,bus=2,addr=5,config=1,if=1,alt=0
> HID++ interface IN endpoint: ep #2 IN (address 0x82) interrupt - undefined usage [20 bytes]
> Dongle Info
> 
> 	Firmware (maj.minor.build):  RQR21.00.B0007
> 	Bootloader (maj.minor):      02.14
> 	WPID:                        8006
> 	(likely) protocol:           0x07
> 	Serial:                      a2:18:9e:70
> 	Connected devices:           0
> 
> Closing Logitech receiver in Firmware mode (not bootloader)...
> 

> ./munifying pair
> Found CU0007 Dongle for G700/G700s mouse
> Using dongle USB config: Configuration 1
> Resetting dongle in order to release it from kernel (connected devices won't be usable)
> EP descr: ep #1 IN (address 0x81) interrupt - undefined usage [8 bytes]
> EP descr: ep #2 IN (address 0x82) interrupt - undefined usage [20 bytes]
> HID++ interface: vid=046d,pid=c531,bus=2,addr=5,config=1,if=1,alt=0
> HID++ interface IN endpoint: ep #2 IN (address 0x82) interrupt - undefined usage [20 bytes]
> Enable pairing for 60 seconds
> USB Report type: HID++ short message, DeviceID: 0xff, SubID: SET REGISTER SHORT, Params: 0xb2 0x00 0x00 0x00
> 	Register address: REGISTER PAIRING
> 	Value: 0x00 0x00 0x00
> ... Enable pairing response (should be enabled)
> 
> Printing follow up reports ...

> LOGITacker (discover) $ pair device run
> Trying to pair using Unifying global pairing address
> <info> LOGITACKER_RADIO: Channel hopping stopped
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: Try to pair new device on target address BB:0A:DC:A5:75
> <info> LOGITACKER_RADIO: Channel hopping stopped
> <info> ESB_ILLEGALMOD: Using channel table 'Unifying pairing'
> <info> ESB_ILLEGALMOD: New channel table with length 11
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase before TX: 0
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase after TX: 1
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  E1 5F 01 DE AD BE EF 82|._......
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  08 13 37 04 00 01 05 0 |......  
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  00 00 00 00 00 A9      |......  
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: TX'ed to BB:0A:DC:A5:75
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: Phase before RX: 1
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:                         |        
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: RX phase after parsing: 2
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase before TX: 2
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase after TX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  E1 40 01 DE 00         |.@...   
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: TX'ed to BB:0A:DC:A5:75
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: Phase before RX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:                         |        
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase before TX: 3
> <warning> LOGITACKER_PROCESSOR_PAIR_DEVICE: update TX payload called for unknown pairing phase: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase after TX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  E1 40 01 DE 00         |.@...   
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: TX'ed to BB:0A:DC:A5:75
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: Phase before RX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:                         |        
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase before TX: 3
> <warning> LOGITACKER_PROCESSOR_PAIR_DEVICE: update TX payload called for unknown pairing phase: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase after TX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  E1 40 01 DE 00         |.@...   
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: TX'ed to BB:0A:DC:A5:75
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: Phase before RX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:                         |        
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase before TX: 3
> <warning> LOGITACKER_PROCESSOR_PAIR_DEVICE: update TX payload called for unknown pairing phase: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: phase after TX: 3
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE:  E1 40 01 DE 00         |.@...   
> <info> LOGITACKER_PROCESSOR_PAIR_DEVICE: TX'ed to BB:0A:DC:A5:75

After executing "pair device run" nothing happens in the muifying pairing process ("Printing follow up reports ..."). I tried this procedure in standard workmode and in g700 workmode and with different OS-setups:

1) Kali VM with Vmware
2) Laptop with pure Kali OS
3) Android and April Dongle connected via OTG

mame82, 2019-12-01:
Pairing request stays unanswered here. Most likely reason: there's already a device paired.

Try `munifying unpairall`

G700 firmware is limited to one paured device

bl4ckOut, 2019-12-03:
Tried it. Unfortunately still the same behaviour.

lixo03, 2020-03-07:
If you are using lightspeed firmware you have to set workmode to lightspeed.
Do this by executing "options global workmode lightspeed"