RoganDawes/LOGITacker #32: Not an issue, just a question

bluegizmo83, 2019-11-18:
Does this current work using a CrazyRadio PA dongle like mousejack? It has an nRF24LU1+ chip. Mousejack is pretty cool and all, but your software here looks to go more in depth and do more stuff.

RoganDawes, 2019-11-18:
Logitacker requires an nRF52840 chip or equivalent, due to its advanced
radio. So far, it only supports actual 52840, although in theory, parts of
it should work well enough on 52832's, for example.

A fairly big part of it currently depends on the USB UART user interface,
though, which is not present on the '32.

bluegizmo83, 2019-11-18:
> Logitacker requires an nRF52840 chip or equivalent, due to its advanced radio. So far, it only supports actual 52840, although in theory, parts of it should work well enough on 52832's, for example. [...]

Oh ok, thanks!

closed 2019-11-18 by bluegizmo83