RoganDawes/LOGITacker #35: dowanload files from target computer

zhjygit, 2019-12-15:
Via tests, I find that I can not download or steal files from target computer ,with no internet network and any other network.
After we get the cmd shell of target computer, I find that the process running on target computer is powershell.exe which started when injected.
Maybe we could improve the framework to upload and download small files between clent and target computer, just like meterpreter framework of others.
I don't know how to finish the work above, anyone can help me???

RoganDawes, 2019-12-15:
It’s a somewhat complicated topic, to be honest. What @mame82 has achieved
making the covert channel entirely self-contained (ie all in the dongle) is
amazing, in comparison to similar tools such as USaBUSe, which require
listeners, stagers, etc to achieve their covert channel.

Obviously, to upload and download files, you really want them to end up on
the USB host machine. To achieve this, you need some software running on
the host to manage selection of the file to upload, and where to save
downloaded files to.

One way to achieve this (which may end up being fairly unintrusive) would
be to implement some sort of ZModem protocol, possibly in the dongle
itself. Then, a command such as “covert_channel xx download remote-file”
could send a suitable message to the remote to retrieve the file, then
initiate a ZModem download that the terminal emulator would respond to that
prompts where to save the file. And similarly for upload.

Worth considering, anyway.

The alternative is to have some more sophisticated software running on the
host, communicating with the usb dongle via the serial port or raw hid
channel, switching it into some sort of machine-friendly interface. It
sounds like a LOT more work.

zhjygit, 2019-12-16:
Thanks very much Sir!
Well,application to application communication is useful. 
I have tried to download and upload files based on stm32+esp8266, without any other network between target computer and client, the rate is about 2.5kbps. 
The way I tried is similar to Zmodem protocal, maybe I should try Zmodel protocal.