RoganDawes/LOGITacker #38: every second injection fails

JamesBohrmann, 2019-12-17:
hello,

i use this script to inject:
LOGITacker (discover) $ script press NUMLOCK
LOGITacker (discover) $ script press GUI R    
LOGITacker (discover) $ script delay 500
LOGITacker (discover) $ script altstring "notepad.exe"
LOGITacker (discover) $ script delay 500
LOGITacker (discover) $ script press RETURN
LOGITacker (discover) $ script altstring "here comes some demo text. Very much characters. At least 24!"

what happens:
the first run works fine, the second run, whichout any changes, fails. 

any ideas what could be wrong?

thanks in advance
James

RoganDawes, 2019-12-19:
Could you try running something simple, like just typing out a string and pressing enter?
Leave notepad open and in focus, so you can see if it works?
And then see if it works every time you trigger it?

That will show whether there is something wrong at a hardware/firmware level, or something wrong with your script. 

Then we can debug from there.

PaulaPaul750, 2020-01-07:
hello,
it works like a charm. so it seems it have something to do with the script, maybe the NUMLOCK thing.

RoganDawes, 2020-01-07:
Keep in mind that NUMLOCK is a toggle. So the first run turns it on, the
second run turns it off. Would definitely explain why your script fails.

More generally, a HID keyboard *should* be notified of the system toggle
states so that it can set its own LEDs accordingly, so it is technically
possible to query this before triggering the toggle. Whether that
functionality exists or not in LOGITacker is a different question.

However, I believe that only Windows and Linux actually do synchronized
toggle states across HID keyboards. OS X apparently does not. Whether this
is a problem for you depends on your targets, I guess!