RoganDawes/LOGITacker #53: feature request: save a bit of keystrokes per device

PlasmaHH, 2020-02-19:
Hey, another thought that might improve things in a noisy environment, maybe its worth implementing. I have honestly no idea for the different supported hardware how much memory is available but might be worth thinking about it.

The idea is to have a ringbuffer of a couple of bytes/keystrokes/mouse movements associated with each device (especially for there being a few unencrypted around here still), maybe configurable at compile/run time? When listing the devices this could then display those, which is a bit nicer in display than the single keycodes in the logs as well as a bit better for many devices than passing it through to the host.

RoganDawes, 2020-02-20:
Sorry, I'm not sure if I follow what you are suggesting. Can you explain in more detail, please?

lkarlslund, 2020-08-17:
I think the suggestion is to buffer a few incoming keystrokes at a time, so you display them after a timeout. So instead of getting many many single keypress logs, you get a concatenated string of keys, when user is idle for some timeout value in milliseconds.

RoganDawes, 2020-08-17:
Hmm, interesting idea.
Keep a buffer of say 8 keystrokes for each observed device, start filling it up, and log the time it was added. If another keystroke comes in, add it to the buffer, and update the time. A timer task periodically checks each per-device buffer timestamp, and if it is more than X period old, prints and clears the buffer. If the buffer gets filled, print and clear immediately.
Probably not too hard to implement, but may require more memory than is available.