RoganDawes/LOGITacker #54: Option to store/load covertchannel payloads

nieldk, 2020-03-10:
I believe it would be fairly simple to change the code for coveetchannel payloads (sharplocker atm is ‘hardcoded’.
Since the payload is executed by a simple powershell base64 encoded string, we could change the function to add options for loading/storing the base64 string, similar to when devices are added/stored.
Practically, that would make LOGITacker an advanced Rubberducky.
I will look into that, if no one do before me ;)

SymbianSyMoh, 2020-03-10:
@nieldk The problem is that you are limited by the flash size of the LOGITacker supported device. Anyways, Covertchannel is just a normal script payload, and using `script` command functions would do the job.

RoganDawes, 2020-03-11:
Normal, apart from the number of lines in the payload ...

Marcus explained on Twitter a number of issues with the proposal, that need
to be carefully considered.

I do wonder, though, if it is possible to change the flash allocation map
to make more flash pages available for storage, at the cost of program
space?