RoganDawes/LOGITacker #61: How can I suppress the output when activating Logitacker?

git-minix, 2021-11-13:
When I am connect with

`$ screen / dev / ttyACM0 19200`

, I am immediately spammed with INFO messages because LOGITacker immediately starts with auto discovery. How can i stop this? You don't even get to use the commands because messages are constantly being output. Seems like a good solution logitacker-script, which I found on youtube (https://www.youtube.com/watch?v=_NgOmR0l1p0&t=72s) - but there is not link ta that script. 

Is there any option to start quite?  Is there an option to do the cli interactions and the output in two different output channels?

Frank

git-minix, 2021-11-13:
guess this helps

`$ log backend m_cli_cdc_acm_log_backend halt`

frame84, 2024-10-16:
```
LOGITacker (discover) $ log disable 
LOGITacker (discover) $ 
```

closed 2024-10-17 by RoganDawes