RoganDawes/LOGITacker #63: Porting LogiTacker console to UART

JohnHonai2255, 2022-12-07:
I was trying to port the Logitacker console into UART. I tried to alter the code in main.c, nrf_cli.h etc, taking reference to a UART example code ([cli](https://github.com/justaddobvious/nRF5_SDK_15.3.0_59ac345/tree/edb62a71cb6b137edccbc1e9ce30b6c3a6f736b6/examples/peripheral/cli) - with small changes) from [nRF5_SDK_15.3.0_59ac345](https://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v15.x.x/). But i was not able to get the uart.

It would be so great, if you could guide me.

RoganDawes, 2022-12-07:
I'm not entirely sure what you are trying to do, and what you have done so far that hasn't worked. Can you elaborate a bit, please?

JohnHonai2255, 2022-12-07:
I would like to integrate Logitacker which is implemented in nRF52840 with another development board. I wanted to access the logitacker shell (which is currently implemented through USB)  through UART in the development board, from the nRF52840.

mame82, 2022-12-07:
There shouldn't be a need to modify SDK headers (`nrf_cli h`), but you have to carefully craft the config files for your custom board, iot include required libs during compilation.

Also there might be a need to use two UARTs iot deconflict 'uart_cli' and 'uart_logging'

Look through my custom board configs for MakerDiary MDK, adjust settings as required and carefully match them to your board:

https://github.com/RoganDawes/LOGITacker/tree/master/mdk/blank/config

Sorry for not being more helpful, still lacking the time to support this

mame82, 2022-12-07:
The `esb_illegalmod` and `cli_mod` folders are copies from the nrf5 SDK, with required changes.

Tbh, I can't remember why I had to do changes to CLI implementation, but they shouldn't be "invasive" (needs diffing, to find out what I did back then).

To be sure, just include the correct libs in the Makefile for your board. Again, the Makerdiary MDK file could serve as example. It includes the `cli_uart` lib (Makefiles for the 3 supported boards don't differ library-wise, but otherwise are tuned to the respective board)

https://github.com/RoganDawes/LOGITacker/blob/master/mdk/blank/armgcc/Makefile#L143

https://github.com/RoganDawes/LOGITacker/blob/master/mdk/blank/armgcc/Makefile#L143

closed 2024-01-01 by RoganDawes