RoganDawes/LOGITacker #64: C-U0012 Unifying Receiver and nRF52840 MDK USB Dongle - Pairing Timeout

TW-D, 2023-04-26:
Hello, after discovering your project in the MISC 126 magazine, I started the adventure but after several efforts I always get a too long delay during the pairing phase.

Here is the detailed procedure I followed :

>
> Installation of "munifying"
>

```
:~$ sudo apt update
:~$ sudo apt-get install golang-go libusb-1.0-0 libusb-1.0-0-dev
:~$ go install github.com/mame82/munifying@latest
```

>
> Flash "Logitech C-U0012 Unifying Receiver" With "RQR39.06_B0040"
>

```
:~/LOGITacker$ wget https://raw.githubusercontent.com/Logitech/fw_updates/master/RQR39/RQR39.06/RQR39.06_B0040.shex

:~/LOGITacker$ lsusb
[...] ID 046d:c52b Logitech, Inc. Unifying Receiver

:~/LOGITacker$ sudo ~/go/bin/munifying flash --hexfile ./RQR39.06_B0040.shex

:~/LOGITacker$ lsusb
[...] ID 046d:c539 Logitech, Inc. USB Receiver
```

>
> Flash "nRF52840 MDK USB Dongle" With "LOGITacker"
>

```
:~/LOGITacker$ wget https://github.com/RoganDawes/LOGITacker/releases/download/v0.2.3-beta/logitacker_mdk_dongle.hex

:~/LOGITacker$ python3 ./uf2conv.py ./logitacker_mdk_dongle.hex --family 0xADA52840
```

>
> "Logitech C-U0012 Unifying Receiver" And "nRF52840 MDK USB Dongle" Pairing
>

```
:~/LOGITacker$ sudo ~/go/bin/munifying info
:~/LOGITacker$ sudo ~/go/bin/munifying unpairall
:~/LOGITacker$ sudo ~/go/bin/munifying pair
TIMEOUT :(
```

__NOTE :__ All these operations have been done under Ubuntu 22.10

Thank you for your help.

TW-D, 2023-05-11:
Finally, I had forgotten to do ```LOGITacker (discover) $ pair device run``` before ```:~/LOGITacker$ sudo ~/go/bin/munifying pair```. Thanks to the author of the article :)

closed 2023-05-11 by TW-D

RoganDawes, 2023-05-11:
Thanks for including the solution!