RoganDawes/LOGITacker #67: Can't connect to com on Android

xxxcrow, 2023-09-18:
Hi, I've got the makerdiary dongle, haven't tested it fully yet
I usually connect with a raspberry or my laptop
And on android i'm using an app "USB serial to telnet" and for this dongle and firmware it drops the connection (although working for some other serial things connected)
Is there any other way I can connect it to android?
Thank you

RoganDawes, 2023-09-18:
Sorry, I haven't tried to use an Android device as a host.

xxxcrow, 2023-09-18:
@RoganDawes I hope it's possible
Esp32 works flawlessly
Are there any serial specific options I can use, because my app settings have some?
It has data bit, stop bits, parity, force local echo off, converted client's CR-LF to CR

RoganDawes, 2023-09-18:
The LOGITacker USB serial implementation is purely USB, with no real use of baud rates, etc that could be incorrect. i.e. if you connect at 9600, you should have exactly the same experience as connecting at 115200 or higher baud rate.
What exactly do you mean by "drops the connection"? Does the serial port disappear? That suggests that the power consumption of the LOGITacker dongle is higher than the Android host is able to tolerate, and it is being disconnected at a USB level. Could you possibly try with a powered USB hub in between the Android host and the LOGITacker dongle?

RoganDawes, 2023-09-18:
Marcus had success with Another Term Lite, it seems: https://twitter.com/mame82/status/1172543785164845058

xxxcrow, 2023-09-18:
@RoganDawes wow, "Another Term" is working 😶‍🌫️ although the app looks hardly usable for me 🥲

RoganDawes, 2023-09-18:
Glad to know it is working. What makes it unusable?

xxxcrow, 2023-09-18:
@RoganDawes did you see it? 😅 it looks like a terminal version of lucky patcher, if you know what that is 😅
The UI, the keyboard, everything looks like android gingerbread (android 2)

RoganDawes, 2023-09-18:
I haven't tried it. I guess the point is that the problem is not with LOGITacker, but with the terminal apps on your host platform. So, not relevant here.

closed 2023-09-18 by RoganDawes