RoganDawes/LOGITacker #69: All devices display almost the same info, and can't do anything with them

xxxcrow, 2023-10-08:
Hi, all devices I find display as:
```
'unknown name' keyboard: no mouse: no                           class: Logitech Unifying compatible device WPID: 0x0000 dongle WPID: 0x0000
```
Or:
```
'unknown name' keyboard: yes mouse: no                           class: Logitech Unifying compatible device WPID: 0x0000 dongle WPID: 0x0000
```
Or:
```
'unknown name' keyboard: no mouse: yes                           class: Logitech Unifying compatible device WPID: 0x0000 dongle WPID: 0x0000
```

Question 
#1: is it ok?
#2: what are these devices?
#3: when I try passive/active enumerate the output is the same for all devices: timeout after 1300ms

RoganDawes, 2023-10-10:
Don't think I have seen this before. Are you sure that you are picking up actual Logitech devices? Could there possibly be other brand wireless devices around you?

xxxcrow, 2023-10-10:
@RoganDawes I don't know if these are real Logitech devices, because I don't have any 
But I travel to a lot of places and everything I find is like that
Sometimes I find none, sometimes 4-5-6
It has more logs, about the correct device, but an empty frame

RoganDawes, 2023-10-10:
Can I suggest buying a cheap Logitech mouse with Unifying receiver to use as a testbed? That way you will know whether you are dealing with a real Logitech device, and what to expect LOGITacker to display when you see other devices in the field.

closed 2023-10-10 by RoganDawes

xxxcrow, 2023-10-10:
@RoganDawes shouldn't non-Logitech devices be filtered? 🤔
The logs say something about correct signature and add the devices to the list
But some get filtered
Doesn't it mean the filter is not working? 🙄