RoganDawes/LOGITacker #71: LNA/PA support for Crazy Radio 2.0

rafale0n, 2024-02-13:
Hey, I love your project. 

I just got this dongle with supported nRF52840 chipset which according to description adds radio power amplifier (PA) providing up to +20dBm output power and LNA. I have noticed that LOGItacker only supports up to +8dBms, not taking full advantage of this device. 

Would it be possible to add support for additional built in power amplifier and LNA? This would greatly improve discovery and communication with vulnerable devices. I would be happy to donate to your project, if you provide such option. 

All the best.
R.

RoganDawes, 2024-02-13:
It sounds great, but I am unlikely to work on this, as I have no hardware to test with.

closed 2024-02-13 by RoganDawes

rafale0n, 2024-02-26:
Thanks for getting back to me. I have one extra spare Crazy Radio 2.0 device, which I will be more than happy to send it to you at my cost, if you are willing to give it a try. Let me know if you are up for it. Fingers crossed though.

Regards,
Raf.

(Sorry for re opening the issue)

RoganDawes, 2024-02-27:
I'm probably not going to be able to do this, I'm afraid.

Regards,

Rogan