RoganDawes/LOGITacker #72: Potential bug concerning active enumeration

frame84, 2024-10-16:
Hi,

I'm posting this message because I'm wondering if there might be a bug with the active enumeration. 

While studying the features that allow LOGITacker to identify if a device is vulnerable to keystroke injection, I became interested in the _logitacker_processor_active_enum.c_ file and more specifically in the _processor_active_enum_esb_handler_func_()_ function. It is indeed in this function that a device is qualified as vulnerable to keystroke injection (line 311 : _p_device->vuln_plain_injection = true;_) depending on the test sequence.

As I'm not a developer, I have serious doubts about my ability to understand the code in depth, but I wonder if there might be a problem with the switch/case of this function:

```
void processor_active_enum_esb_handler_func_(logitacker_processor_active_enum_ctx_t *self, nrf_esb_evt_t *p_esb_event) {
    ...   
    switch (p_esb_event->evt_id) {
        case NRF_ESB_EVENT_TX_FAILED:
        {
            ....
            break;
        }
        case NRF_ESB_EVENT_TX_SUCCESS_ACK_PAY:
        case NRF_ESB_EVENT_TX_SUCCESS:
        {
            ...
            if (self->inner_loop_count == 0) processor_active_enum_add_device_address_to_list(self);
            ....
            self->inner_loop_count++;

            processor_active_enum_update_tx_payload(self);

            if (p_esb_event->evt_id == NRF_ESB_EVENT_TX_SUCCESS_ACK_PAY) {
                ....
                while (nrf_esb_read_rx_payload(&self->tmp_rx_payload) == NRF_SUCCESS) {
                    ....
                    if (rf_report_type == UNIFYING_RF_REPORT_LED) {
                        ....
                        NRF_LOG_INFO("ATTACK VECTOR: devices accepts plain keystroke injection (LED test succeeded)");
                        ...
                        if (p_device != NULL) {
                            p_device->vuln_plain_injection = true;
                            ....
                            }
                        }
                    } else if (rf_report_type == UNIFYING_RF_REPORT_PAIRING && device_id == PAIRING_REQ_MARKER_BYTE) { 
                                ....
                            }
                        }
                    }
                    ...
                }
            }
            ....
            break;
        }
        case NRF_ESB_EVENT_RX_RECEIVED:
        {
            ....
            break;
        }
    }
    ...
}
```

I think there could be a problem with the case where `p_esb_event->evt_id` has the value `NRF_ESB_EVENT_TX_SUCCESS_ACK_PAY`. Unless I'm mistaken, the following condition is never true (line 296):

```
            if (p_esb_event->evt_id == NRF_ESB_EVENT_TX_SUCCESS_ACK_PAY) {
                ....
            }
```

Unless the call to the `processor_active_enum_update_tx_payload(self);` function on line 292 could modify the value of `p_esb_event->evt_id`?

If that's not the case, then I can't see how the LED test can be triggered, and the device identified as vulnerable.

In practice, I've tried using the active enumeration feature on an unencrypted mouse, which is vulnerable to keystroke injection. LOGITacker didn't recognize the dongle as vulnerable (the device remained blue, not green, when the `devices` command was run).

I'm trying to understand why this test fails.

Thank you for your clarification and thanks again for this amazing project !!

RoganDawes, 2024-10-17:
I'm afraid this project is not maintained, and since I am not the original author, I'm not able to comment on how it is supposed to work.

closed 2024-10-17 by RoganDawes

frame84, 2024-10-17:
Thanks Rogan for the feedback, I understand. I'll try to dig the subject on my own and come back here if there's something interesting to share.